WeActStudio/MiniSTM32H7xx #5: Test projects compiled  with GCC doen't work properly

WiseLord, 2020-09-25:
I purchased the board and tried to compile the examples under GCC arm-none-eabi. I converted the following two projects into 
"Makefile" project type via STM32CubeMX:
- SDK/HAL/03-LCD_Test/03-LCD_Test.ioc
- SDK/HAL/08-DCMI2LCD/08-DCMI2LCD.ioc

Than, I added missing Drivers/BSP/ST7735/*.c files into Makefile, as well as missing defines (TFT96) found in .uvprojx

My results:

- 03LCD_Test compiled with gcc-arm-none-eabi <= 7 works, while one compiled with gcc-arm-none-eabi >= 8 doesn't work. I can see brightness changing (so timer PWM is OK), but display initialization is failed whwn recent GCC is used.

- 08-DCMI2LCD has the same situation regarding LCD, while camera doesn't work in any case (it's detected, but nothing is shown on the screen even with "good" compiler).

So, the question: have the developers ever tried to build their examples in something different than Keil? It would be wonderful to have the examples really OS- and IDE-independent.

WeActTC, 2020-09-25:
1.The compiler overoptimizes 2.just going to step through and see what's wrong with the code 3.This is a bug in our code, need to optimize

WiseLord, 2020-09-25:
Actually, I also tried to disable optimization at all, with no effect. When I have some more free time, I'll try to look through it again.

On the other hand, you could also try recent GCC on your side.

WeActTC, 2020-09-25:
OK, If I have time, I will try!

Liuyufanlyf, 2021-03-08:
> I purchased the board and tried to compile the examples under GCC arm-none-eabi. I converted the following two projects into
> "Makefile" project type via STM32CubeMX:
>
> * [...]

I believe [the newest pull request](https://github.com/WeActTC/MiniSTM32H7xx/pull/13) has solved this problem, you can have a try!

closed 2022-07-24 by WeActStudio