WeActStudio/MiniSTM32H7xx #15: ST7735 1.8inch LCD 驱动代码

Liuyufanlyf, 2021-04-15:
觉得 1.8inch LCD 的驱动代码真的有必要重构一下，现在是在 ST 的 ST7735 驱动上面包了一层 LCD 的封装，绘制点线多边形这些函数是 ST7735 层级的，显示字符串却是 LCD 层级的，亮度调整则两边都有，字符和字符串的显示也只限定了一种画笔色和背景色，初始化函数也没有包装出来。总之用户拿到手以后还要花很多力气去修改。个人建议要么就封装一个完整的 LCD 层，要么就拓展 ST7735 的驱动，LCD 层只用于配置。

anthroid, 2021-09-18:
I fully agree. Also, there is another ST7735 implementation that appears to address some of these problems [here](https://github.com/afiskon/stm32-st7735).

emalagonca, 2021-12-23:
The code is only an example to get started. It seems to be a bad idea to try to use the code in your project, the next piece of code was taken from the LDC_Light function, it contains some curious code like this temp2 = time_now - 0; there are more things to improve but this is only an example, so what's the problem?. I like this project for the hardware not for the software.

	temp1 = Brightness_Now;
	temp1 = temp1 - Brightness_Dis;
	temp2 = time_now;
	temp2 = temp2 - time;
	
	k = temp1 / temp2;
	
	uint32_t tick=get_tick();
	while(1)
	{
		delay_ms(1);
		time_now = get_tick()-tick;
		temp2 = time_now - 0;
		set = temp2*k + Brightness_Now;
		LCD_SetBrightness((uint32_t)set);
		if(time_now >= time) break;		
	}

closed 2022-07-24 by WeActStudio