WeActStudio/MiniSTM32H7xx #23: LCD orientation

jongwoon-choi, 2021-11-16:
I want to use LCD as Landscape mode.

I found the "03-LCD_Test" project is set up as Portrait mode, it is not easy for me to change to Landscape mode.
Would you tell me how to change from Portrait to Landscape mode in "03-LCD_Test" project?

Best regards,
Jongwoon

WeActStudio, 2022-07-24:
You can read ST7735.c, and read an array of OrientationTab

closed 2022-07-24 by WeActStudio

jongwoon-choi, 2022-07-24:
p{margin-top:0;margin-bottom:0}
    

    Thanks for your answer.
--------- 원본 메일 ---------

    
        보낸사람: WeAct Studio ***@***.***>
        받는사람: WeActStudio/MiniSTM32H7xx ***@***.***>
        
            참조: Jongwoon Choi ***@***.***>, Author ***@***.***>
        
        날짜: 22.07.24 23:22 GMT +0900
        제목: Re: [WeActStudio/MiniSTM32H7xx] LCD orientation (Issue #23)
        
        
    
    
        
Closed #23 as completed.