WeActStudio/MiniSTM32H7xx #25: STM32H743 pinout

S0mbre, 2022-02-03:
Made a **colorful pinout** for the **STM32H743VIT board** using the official WeAct schemes and STM32 datasheet.

Please review and if OK, I will send updated version soon (with all AF's added to pins)
[STM32H7 Pin Layout.pdf](https://github.com/WeActTC/MiniSTM32H7xx/files/7992174/STM32H7.Pin.Layout.pdf)
[stm32h743 pinout.zip](https://github.com/WeActTC/MiniSTM32H7xx/files/7992176/stm32h743.pinout.zip)
.

ctag, 2023-05-27:
Hi, the files links lead to 404.

S0mbre, 2023-06-01:
Updated, please find attached here.
[STM32H7 Pin Layout.pdf](https://github.com/WeActStudio/MiniSTM32H7xx/files/11630198/STM32H7.Pin.Layout.pdf)
[stm32h743 pinout.zip](https://github.com/WeActStudio/MiniSTM32H7xx/files/11630199/stm32h743.pinout.zip)

skimans, 2025-03-28:
hi @S0mbre 
there is update of pinout ?
thanks