WeActStudio/MiniSTM32H7xx #31: 12-Thermopile-Shield Openmv examples does not works

Kabron287, 2022-07-25:
I used PB10 as SCL and PB11 as SDA signals for MLX90640
I see clock and data signals on scope at 50kHz when fir.init() executed, but then I got  error message:
**ValueError: Failed to detect a supported FIR sensor.** 
These examples works fine with the original Openmv H7 board.

Kabron287, 2022-07-25:
1. It is definite bug in current FW. Look at the screenshot, MLX90640 address is found correctly, but then nothing happened.
2. There is no definition for **FIR_MLX90640** at all in FW.
Extended command **fir.init(type = fir.FIR_MLX90640)** allow to skip address searching, but in current FW version I got:
**AttributeError: 'module' object has no attribute 'FIR_MLX90640'** error

[image]
.

radiohound, 2023-01-31:
I am interested to find a solution for this as well. Thanks Kabron287 for your trouble shooting.