WeActStudio/MiniSTM32H7xx #43: Can I get information about the use of PINs?

rizacelik, 2023-11-10:
31 PINs are reserved for hardware such as camera, LCD screen, MicroSD, SOUND. There are very few PINs left.
Can 31 PINs be used for other things? So, if not all, at least some of them can be used? If it is not used, why did you give it on the board?.
I use your boards for Flight Controller and share the Firmware and installations. I would like to use this motherboard, but I wanted to ask because it has very few PINs.

synersignart, 2024-02-15:
so far I have few info on it , the micro sd card uses the SDIO making SPI3 unusable , one of the  SPI1 is occupied by the Flash but I have no idea how to make it work and the LCD is sitting on SPI4 so this is unusable .  so the only usable is SPI2 for the IMU how ever I2C 1 and 2 are free

rizacelik, 2024-02-21:
Thank you for your response. I gave up on the WeAct Studio H743 card. I ordered the devEBOX H743 board instead. I'll work on it when it arrives.

rizacelik, 2024-02-21:
I waited for an answer for a long time, but it was not answered. That's why I gave up.  Thank you.

lorenzoridolfi, 2024-10-20:
I won't buy this board anymore due lack of basic support.