WeActStudio/MiniSTM32H7xx #49: New board definitions for OpenMV v4.4.1 or v4.5.5

sashaegorov, 2024-07-19:
Hi, **Dear WeActStudio Team**!

First of all, thank you so much for great products! I'm happy owner of two WeAct STM32 H743 boards, Mini STLink, USB to TTL converter, CH9143 BLE to USB2UART, WeAct ESP32 few others products. They are absolutely awesome!👍🏻 I'm big fan how well these board built and designed.

I think this great board, and STM32H742 is promising MCU, it will be my to-go board for long time. However, I think maintenance of SDK could be improved, along with the documentation. I can help with testing, building or writing examples which is very helpful for new users, who don't know C/C++ language; or contribute something to this project.

Regarding WeAct H743 board I'm mostly aimed on it's OpenMV and MicroPython (not checked yet) capabilities. Currently I'm trying to build the latest version of OpenMV (which is [v4.5.5](https://github.com/openmv/openmv/tree/v4.5.5)) and it's failing. It has a lot of compiler warnings and errors with no quick fix.

If you have definitions you used to compile [SDK/openmv/Firmwares/V4.4.1](https://github.com/WeActStudio/MiniSTM32H7xx/tree/master/SDK/openmv/Firmwares/V4.4.1) firmware, can you push to this repo? I'll try to compile them with v4.5.5

I've checked the following:
- [SDK/openmv/Firmwares/V4.4.1](https://github.com/WeActStudio/MiniSTM32H7xx/tree/master/SDK/openmv/Firmwares/V4.4.1) - updated one year ago or so, however it works well with the board👍🏻
- Board definitions
  - [SDK/openmv/Ports/omv/boards/WeActStudioSTM32H7xx](https://github.com/WeActStudio/MiniSTM32H7xx/tree/master/SDK/openmv/Ports/omv/boards/WeActStudioSTM32H7xx) I've tried to compile with this board definition, it for [openmv firmware to v3.9.3](https://github.com/WeActStudio/MiniSTM32H7xx/commit/5998d6307fe56e66c1dd2178a1d292eb68e86692) only
  - [SDK/openmv/Ports/micropython/boards/WeActStudioSTM32H7xx](https://github.com/WeActStudio/MiniSTM32H7xx/tree/master/SDK/openmv/Ports/micropython/boards/WeActStudioSTM32H7xx) same story here
- WeActStudio Repos
  - [WeActStudio/openmv](https://github.com/WeActStudio/openmv) repo, looks like it was updated long time ago
  - [WeActStudio/micropython](https://github.com/WeActStudio/micropython) same, updated long time ago

sashaegorov, 2024-10-23:
@WeActStudio please🙏

WeActStudio, 2024-10-24:
Need in this branches.

---- Replied Message ----
| From | Alexander ***@***.***> |
| Date | 10/24/2024 04:07 |
| To | ***@***.***> |
| Cc | WeAct ***@***.***>***@***.***> |
| Subject | Re: [WeActStudio/MiniSTM32H7xx] New board definitions for OpenMV v4.4.1 or v4.5.5 (Issue #49) |

@WeActStudio please🙏